NModbus4/NModbus4 #14: Read/Write signed value

AnthoSi, 2015-04-13:
For my needs, i have to write/read signed values. But NModbus seems to only use ushort data type for input and output. Have to create a new branche ?

Maxwe11, 2015-04-13:
You can convert signed values to unsigned

AnthoSi, 2015-04-13:
I already did that, but the value written to my controller must be signed. It's not the case if i convert it to unsigned value..

Maxwe11, 2015-04-13:
@AnthoSi everything is bytes :smile: Difference between signed and unsigned in that how we interpret that bytes. 
Convert signed value to unsigned - write it to controller - convert back to signed on controller side.

AnthoSi, 2015-04-13:
I know, but i can't change anything on the controller. Why don't use short data type instead of ushort ?

Maxwe11, 2015-04-13:
> Why don't use short data type instead of ushort ?

Why do we need to do so?

>I know, but i can't change anything on the controller. 

So you can control `ushort`s that you write.

AnthoSi, 2015-04-13:
Sorry for my lack of knowledge about Modbus. i solved my problem.
Thanks

closed 2015-04-13 by Maxwe11