NModbus4/NModbus4 #25: Portable subset support

Maxwe11, 2015-06-03:
For making NModbus4 portable we should split up abstractions from transport implementation.
I'd suggest to split up on following packages:
1. NModbus4.dll - portable abstractions
2. NModbus4.TcpUdp.dll - portable Tcp/Udp transport implementation (depends NModbus4.dll)
3. NModbus4.Serial.dll - non-portable serial  transport implementation (depends NModbus4.dll)

These changes will require some breaking changes e.g. we will have to delete some factory methods like:
```C#
public static ModbusIpMaster CreateIp(TcpClient tcpClient)
public static ModbusIpMaster CreateIp(UdpClient udpClient)
public static ModbusIpMaster CreateIp(SerialPort serialPort)
```
since they require arguments of types that not supported in portable subset. And introduce new factory for creating masters/slaves with corresponded transport.
For making TCP/UDP transport portable we also will have to reimplement it to rely directly on `Socket` instead of `TcpListener`, `TcpClient`, `UdpClient, etc.

breyed, 2015-06-04:
Sounds like a good plan. The breaking changes should be easy enough to deal with. One consideration: If most uses of the library are with IP or if the IP-specific code is small, it would be simpler to include the IP-specific code into NModbus4.dll. Loading one DLL is faster than loading two, and it's less to manage. If you do keep a separate DLL for IP, I'd call it NModbus4.IP.dll, rather than spell out TcpUdp.

If you stick with just two DLLs, it also means you can have just two NuGet packages, one for each DLL: NModbus4 and NModbus4.Serial. Only people who need serial would get the latter, which would depend of the former.

Maxwe11, 2015-06-04:
Agreed :+1:

Maxwe11, 2016-02-01:
Resolved with #86

closed 2016-02-01 by Maxwe11