NModbus4/NModbus4 #57: Integration tests periodically fail

Maxwe11, 2015-08-19:
Discovering: NModbus4.IntegrationTests
	Discovered:  NModbus4.IntegrationTests
	Starting: NModbus4.IntegrationTests.DLL
	System.ArgumentException: EndPoint 127.0.0.1:1060 cannot be removed, it does not exist.
	   at Modbus.Device.ModbusTcpSlave.OnMasterConnectionClosedHandler(Object sender, TcpConnectionEventArgs e) in C:\projects\nmodbus4-ss8e4\NModbus4\Device\ModbusTcpSlave.cs:line 153
	   at Modbus.Unme.Common.EventUtility.Raise[T](EventHandler`1 handler, Object sender, T e) in C:\projects\nmodbus4-ss8e4\NModbus4\Unme.Common\EventUtility.cs:line 24
	   at Modbus.Device.ModbusMasterTcpConnection.<>c.<ReadHeaderCompleted>b__19_0(ModbusMasterTcpConnection thisRef, IAsyncResult asyncResult) in C:\projects\nmodbus4-ss8e4\NModbus4\Device\ModbusMasterTcpConnection.cs:line 88
	   at Modbus.Device.ModbusMasterTcpConnection.CatchExceptionAndRemoveMasterEndPoint(IAsyncResult ar, Action`2 action, String endPoint) in C:\projects\nmodbus4-ss8e4\NModbus4\Device\ModbusMasterTcpConnection.cs:line 172
	   at Modbus.Device.ModbusMasterTcpConnection.ReadHeaderCompleted(IAsyncResult ar) in C:\projects\nmodbus4-ss8e4\NModbus4\Device\ModbusMasterTcpConnection.cs:line 82
	   at System.Net.LazyAsyncResult.Complete(IntPtr userToken)
	   at System.Net.ContextAwareResult.CompleteCallback(Object state)
	   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
	   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
	   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
	   at System.Net.ContextAwareResult.Complete(IntPtr userToken)
	   at System.Net.LazyAsyncResult.ProtectedInvokeCallback(Object result, IntPtr userToken)
	   at System.Net.Sockets.BaseOverlappedAsyncResult.CompletionPortCallback(UInt32 errorCode, UInt32 numBytes, NativeOverlapped* nativeOverlapped)
	   at System.Threading._IOCompletionCallback.PerformIOCompletionCallback(UInt32 errorCode, UInt32 numBytes, NativeOverlapped* pOVERLAP)
	Command exited with code 1

https://ci.appveyor.com/project/Maxwe11/nmodbus4-ss8e4/build/portable-3.0-67

Maxwe11, 2015-08-19:
/cc @breyed @richardlawley have you ever seen this before?

richardlawley, 2015-08-19:
I had a brief go at these earlier and most were due to running tests in
parallel (I use ncrunch) but I was not able to get them to consistently
pass either. Was going to wait until they were merged before looking
properly.