NModbus4/NModbus4 #72: Error codes of failed requests

FonyRockfeel, 2015-09-12:
How to get an error code of failed requests? It is necessary for my project.

FonyRockfeel, 2015-09-12:
I can't find the way to get these codes i source code of framework..

Maxwe11, 2015-09-12:
@FonyRockfeel could you clarify what error code do you need? 
If you mean modbus exception codes then `SlaveException`  and its  property `SlaveExceptionCode` is what you need.

FonyRockfeel, 2015-09-14:
I have this code:

        PortName = cfg.Port;
            
         _port = new SerialPort(PortName) {BaudRate = cfg.BaudRate, ReadTimeout = cfg.Timeout};

         _modbus = ModbusSerialMaster.CreateRtu(_port);
           
         _modbus.Transport.Retries = 3;
         _modbus.Transport.WaitToRetryMilliseconds = 250;
         var regs = _modbus.ReadHoldingRegisters(slave, start, amount).ToList();

How to use SlaveException class in this code?

Maxwe11, 2015-09-14:
`ReadHoldingRegisters` throws `SlaveException` in case if your device returns modbus exception code

closed 2015-09-14 by FonyRockfeel