NModbus4/NModbus4 #75: Initial UWP support

Maxwe11, 2015-09-24:


sandord, 2016-07-10:
Any news on this? The TCP transport in particular...

Maxwe11, 2016-07-11:
@sandord could you try 3.0-alpha2 package?

sandord, 2016-07-12:
I hadn't tried the alpha package yet. When I add it to my UWP app (Windows 10 IoT on a RPI3) I get this:

	System.Reflection.Emit.ILGeneration 4.0.1 provides a compile-time reference assembly for System.Reflection.Emit.ILGeneration on UAP,Version=v10.0, but there is no run-time assembly compatible with win10-arm-aot.
	One or more packages are incompatible with UAP,Version=v10.0 (win10-arm-aot).
	System.Reflection.Emit.ILGeneration 4.0.1 provides a compile-time reference assembly for System.Reflection.Emit.ILGeneration on UAP,Version=v10.0, but there is no run-time assembly compatible with win10-x64-aot.
	One or more packages are incompatible with UAP,Version=v10.0 (win10-x64-aot).
	System.Reflection.Emit.ILGeneration 4.0.1 provides a compile-time reference assembly for System.Reflection.Emit.ILGeneration on UAP,Version=v10.0, but there is no run-time assembly compatible with win10-x86-aot.
	One or more packages are incompatible with UAP,Version=v10.0 (win10-x86-aot).

sandord, 2016-07-12:
I just removed the `-aot` runtimes from the `project.json` file and now I can add the package succesfully. I don't have time to do any testing right now but I will follow up when I have any new feedback.

sandord, 2016-07-12:
I quickly tried the samples project and it seems that the netstandard 1.3 version of `System.Net.Sockets` doesn't provide a `Connect` member for `UdpClient`. Perhaps this is due to an UWP restriction and I need to look at `Windows.Networking.Sockets` instead?

hubert-associates, 2016-10-24:
Looking for/at this with iot-core serial/USB. Thanks!

Puiu, 2017-03-30:
Any news on this one? :)

shaggygi, 2017-03-30:
[System.IO.Ports](https://github.com/dotnet/corefx/tree/master/src/System.IO.Ports) was recently added to be .NET Core 2.0.  It only works for Windows and full .NET Framework for now.  [.NET Core 2.1](https://github.com/dotnet/corefx/milestone/12) is supposed to be released later in 2017 that will focus on UWP.  Hopefully, System.IO.Ports will be updated to work with UWP and Linux, as well.