NModbus4/NModbus4 #78: Holding registers do not change after writing in them!

disklosr, 2015-10-17:
Create a slave modbus device.
Send a write command to write into one of the Holding Registers inside the slave
Read the written holding register
The read value is always zero

Maxwe11, 2015-10-17:
@disklosr what version do you use?

disklosr, 2015-10-17:
I used the lastest available beta version from nuget

Maxwe11, 2015-10-17:
@disklosr could you share your code for repro?

disklosr, 2015-10-19:
Turned out it actually writes starting from the next register.

So I issue a command to write value X in holding register 2274
After reading I print the content of register 2274 and the value is 0
But if I print the value of the register 2275 it prints the value X

Is this normal behavior? I am using the TCP implementation of modbus btw.

Maxwe11, 2015-10-21:
Sorry for the delay. Yes it's normal behavior.

Maxwe11, 2015-10-25:
@disklosr if you have any further comments about this feel free to reopen issue.

closed 2015-10-25 by Maxwe11

disklosr, 2015-10-25:
Thanks I will.
I'm actually using your library to make a GUI modbus slave simulator for Windows. Keep up the good work!