NModbus4/NModbus4 #80: Enron Modbus Didn't Work.

monyetsoft, 2015-11-23:
Hello, I've been using modbus from when it's hosted on google code, and everything works great until my last project.

My last project uses Enron Modbus, And I've noticed that in Modbus.Extensions there is already a class to read single 32 bit register.

However, when I've used the extension procedure  "ReadHoldingRegisters32", It didn't work. 

Apparently "ReadHoldingRegisters32" uses the approach of just doing 2 request of 16 bit registers. Well, Enron Modbus doesn't work that way. 

The returned response is a contiguous longer response for just one request. 

I've debugged NModbus and found that the culprit is at 

ModbusIpTransport.cs

internal static byte[] ReadRequestResponse(IStreamResource streamResource) 

specifically the line 

var frameLength = (ushort)IPAddress.HostToNetworkOrder(BitConverter.ToInt16(mbapHeader, 4));

it keep gives off framelength = 3 while after testing, I found out that the correct framelength is actually 7.

after some crude hacking, I've managed to get it to work, although I also have to change ValidateResponse (ReadHoldingInputRegistersRequest.cs) as well, because expectedbytecount also changes (from 2 to 4) and lastly PerformReadRegisters at ModbusMaster.cs so that I got 2 registers of ushort ( I resort to using just "ReadHoldingRegisters" requesting 1 point , instead of ReadHoldingRegisters32)

I know this is very crude hacking, so I'm sorry I can't submit my changes to this project. Just letting you know that there's an issue with enron modbus.

Maxwe11, 2015-12-07:
@monyetsoft thanks for the report, I will think about that. Could you share reference to Enron Modbus spec with details?

monyetsoft, 2015-12-08:
Don't worry about the answering delay mate :)

Anyway, the reference to enron modbus detail I also got from the internet (http://simplymodbus.ca/enron_FC03.htm)

while for testing with enron tcp/serial slave, I used the peakHMI program (you can download it, it's free at http://peakhmi.com/)

the full download link is http://www.hmisys.com/downloads/PeakHMISlaveSimulatorInstall.exe

I already tried it on a physical device too (Dynamics Flow Computer E-Chart).

And one last thing : The stuff that I said didn't work is on Modbus TCP. I didn't test it on the serial protocol. There's a good chance that the NModbus Enron Serial worked without modification.