NModbus4/NModbus4 #81: Read value from specific Modbus address using RTU

Ankita-XIT, 2015-12-03:
Hello,

I am working on a project in which I want to read value for particular Modbus address using RTU protocol. I have checked the code for NModbus 1.11.0.0 Source but did not find any method to read values using RTU. Does it support to read values? How can I read values? 

Moreover, is there any latest version of Modbus?

Thanks,
ankita

Maxwe11, 2015-12-03:
e.g.
```c#
var master = ModbusSerialMaster.CreateRtu(port);
ushort[] registers = master.ReadHoldingRegisters(slaveId, startAddress, 2);
```

Ankita-XIT, 2015-12-03:
Thank you for your response. But I am not getting the actual result.
Here is my code:
<pre>
using (SerialPort port = new SerialPort("COM6"))
			{
				// configure serial port
				port.BaudRate = 115200;
				port.DataBits = 8;
				port.Parity = Parity.None;
				port.StopBits = StopBits.Two;
				port.Open();

				// create modbus master
				IModbusSerialMaster master = ModbusSerialMaster.CreateRtu(port);

				byte slaveId = 1;
                ushort startAddress = 40023;
				//ushort[] registers = new ushort[] { 15};

				// write three registers
				//master.WriteMultipleRegisters(slaveId, startAddress, registers);
                 master.ReadHoldingRegisters(slaveId, startAddress, 1);
			}

Is there anything wrong? Please suggest.

</pre>
Here is my result:
[image]

Maxwe11, 2015-12-03:
I'd suggest you to try check with `Modbus Poll` tool.

Ankita-XIT, 2015-12-03:
Can you please advise what I am doing wrong in above code with Address 40023?

closed 2015-12-04 by Maxwe11