NModbus4/NModbus4 #89: SerialRTU Anomaly

beerygaz, 2015-12-22:
When sending data manually to query a coil, I get the expected result, but from NModbus4 I get a 0x3F result:

Using serial port monitor:
```
[22/12/2015 12:11:33] - Open port COM5 (C:\Program Files\Eltima Software\Serial Port Monitor\SerialMonitor.exe) 
 
[22/12/2015 12:11:41] Written data (COM5) 
    03 01 00 00 00 01 fc 28                           ......ü(         
[22/12/2015 12:11:41] Read data (COM5) 
    03 01 01 01 91 f0                                 ....‘ð           
[22/12/2015 12:14:21] - Close port COM5 
```

Using NModbus4
```
[22/12/2015 12:14:25] - Open port COM5 (C:\Users\gavin\Source\Repos\HSPI_MODBUS\Modbus Plugin\ConsoleApplication1\bin\Debug\ModbusConsole.vshost.exe) 
 
[22/12/2015 12:14:25] Written data (COM5) 
    03 01 00 00 00 01 fc 28                           ......ü(         
[22/12/2015 12:14:25] Read data (COM5) 
    00 3f                                             .?               
[22/12/2015 12:14:28] - Close port COM5 
```

Code:
            Dim port = New SerialPort("COM5", 19200, Parity.Even, 8, StopBits.One)
            port.Open()
            Dim master As ModbusSerialMaster = ModbusSerialMaster.CreateRtu(port)
            master.Transport.Retries = 0
            master.Transport.ReadTimeout = 3000
            Dim coils() = master.ReadCoils(3, 0, 1)

COM port parameters identical.  Any ideas?

Maxwe11, 2016-01-04:
Are you still hitting this issue?

beerygaz, 2016-01-04:
Resolved the issue by removing a terminating resistor on the bus that seemed to be causing random CRC errors.

closed 2016-01-04 by Maxwe11