NModbus4/NModbus4 #91: Connecting as a modbus TCP/IP Client with specification of the node for the slave

MSUKPR, 2016-01-22:
Hi, 

I haven't seen the option how I can connect to the network of modbus devices with one modbus TCP/IP <> RTU converter. 

I have one IP adress, but in order to get data from different devices I need to specify the nodeId of the slave. 

Any hint?

Maxwe11, 2016-01-22:
If you want to use RTU over TCP/IP you shoud create `ModbusSerialMaster` with `TcpClient`.

MSUKPR, 2016-01-22:
Thank you for your fast response. I will check this in a moment

Kind regards,

Mateusz Siatrak
Software Engineer
UK Power Reserve Ltd
Mobile: +44 7525 796501
Email: Mateusz.Siatrak@ukpowerreserve.com

From: Dmitry Turin <notifications@github.com<mailto:notifications@github.com>>
Reply-To: NModbus4/NModbus4 <reply@reply.github.com<mailto:reply@reply.github.com>>
Date: Friday, 22 January 2016 at 11:12
To: NModbus4/NModbus4 <NModbus4@noreply.github.com<mailto:NModbus4@noreply.github.com>>
Cc: Mateusz Siatrak <Mateusz.Siatrak@ukpowerreserve.com<mailto:Mateusz.Siatrak@ukpowerreserve.com>>
Subject: Re: [NModbus4] Connecting as a modbus TCP/IP Client with specification of the node for the slave (#91)

If you want to use RTU over TCP/IP you shoud create ModbusSerialMaster with TcpClient.

MSUKPR, 2016-01-25:
I have again question:

I've got code like that:

using Modbus.Device;
using System;
using System.Net.Sockets;

namespace ModbusRTU_TCP
{
    internal class Program
    {
        private static void Main(string[] args)
        {
            using (TcpClient client = new TcpClient("192.168.250.111", 502))
            {

                IModbusSerialMaster master = ModbusSerialMaster.CreateRtu(client);
                ushort[] test = master.ReadInputRegisters(1, 10, 10);
               
                Console.WriteLine(test)
                Console.ReadLine();
            }
        }
    }
}

But I am not able to get any data, and I'm not getting any errors. Am I missing something?

Maxwe11, 2016-01-25:
1. try set timeout for `TcpClient`.
2. Are you sure that your slave also uses RTU?

closed 2016-01-26 by Maxwe11

ravigyanvihar, 2017-03-15:
@Maxwe11
Sir 
I am facing the same issue can you tell me the above method is correct

juspolo, 2019-02-20:
@Maxwe11 @ravigyanvihar @MSUKPR , for my experience you should use the ModbusIpMaster when using TCP, while you should use ModbusSerialMaster when using Serial :)

For example, if you are using an ethernet to 485 converter you could do something like this:
```
using (TcpClient client = new TcpClient(TargetConverterIP, TargetIPPort)    // e.g. 192.168.1.128 and 502.
{
    // create modbus master
    IModbusMaster master = ModbusIpMaster.CreateIp(client);
    ushort[] holding_register = master.ReadHoldingRegisters(slaveAddress, startAddress, numberOfRegisters);    // e.g. 1, 0, 1.
}
```
I hope that it could help you, even if it is a bit late.