NModbus4/NModbus4 #100: Server Nmodbus4 don't send TCP FIN after client is disconnected

AurelienC, 2016-04-05:
Hello ,
I have a problem with NModbus Library.

When my client is disconnected :
```
CLIENT           SERVER
FIN, ACK  ---->
          <----  ACK
```
and nothing else...!

The server don't send FIN to client.

After timeout, client send RST to server.

And when client try to reconnect, server send a bad SEQ number.

```
CLIENT           SERVER
SYN seq0  ---->
          <----  ACK seq1 (Wireshark said TCP ACKed unseen segment)
RST seq0  ---->
```

Result, just the first connection after the start of server is done.

Please, do you have any idea ?

My code 
```
class ModbusServer
    {
        #region Variables
        private Sql sql;
        private int port;
        TcpListener tcp;
        ModbusSlave slave;

        #endregion

        #region Constructeur
        public ModbusServer(int port = 502)
        {
            this.port = port;

            IPHostEntry ipEntry = Dns.GetHostEntry(Dns.GetHostName());
            IPAddress[] addr = ipEntry.AddressList;
            tcp = new TcpListener(addr[0], port);
            slave = ModbusTcpSlave.CreateTcp(255, tcp);
            slave.DataStore = DataStoreFactory.CreateDefaultDataStore();

            slave.DataStore.DataStoreWrittenTo += new EventHandler<DataStoreEventArgs>(DataStore_DataStoreWrittenTo);
        }

        #endregion

        #region Public
        public void start()
        {
            try
            {

                slave.Listen();
                Console.WriteLine("[MODBUS] Server started");
                Console.WriteLine("[MODBUS] Port " + port);

            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.Message);

            }

        }
        #endregion

        #region Private
        private void DataStore_DataStoreWrittenTo(object sender, DataStoreEventArgs e)
        {
            Console.WriteLine("[MODBUS] Reception of values");

            int iAddress = e.StartAddress;
            List<ushort> data = new List<ushort>();

            if (e.ModbusDataType == ModbusDataType.HoldingRegister)
            {

                for (int i = 0; i < e.Data.B.Count; i++)
                {
                    data.Add(e.Data.B[i]);
                    Console.WriteLine("[" + i + "] " + e.Data.B[i]);
                }

            }
        }
        #endregion
    }
```

Thanks !
Aurélien
from France

Maxwe11, 2016-04-05:
Could you try to `Dispose` slave instance?

AurelienC, 2016-04-05:
Thank you for your response.

Oh, since I use NModbus version 4, my PLC can't connect to the server...

```
PLC      SERVER       502 → 502 [SYN] Seq=0 Win=1024 Len=0 MSS=1024
SERVER   PLC          502 → 502 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
```

I don't know how I can proceed...

` public ModbusServer(int port = 502)
        {

            this.port = port;
            IPAddress address = new IPAddress(new byte[] { 127, 0, 0, 1 });

            tcp = new TcpListener(address, port);

            slave = ModbusTcpSlave.CreateTcp(255, tcp);

            slave.DataStore = DataStoreFactory.CreateDefaultDataStore();
            

            slave.DataStore.DataStoreWrittenTo += new EventHandler<DataStoreEventArgs>(DataStore_DataStoreWrittenTo);

            tcp.Start();
            slave.ListenAsync();

        }`

AurelienC, 2016-04-05:
When the connection is OK :
[WireShark](http://uploads.im/omOLI.png)

james9837, 2016-04-28:
@AurelienC 
In the first code example you gave (4 messages up), I don't see a call to 'tcp.Start()'.

Later you mention using NModbus 4 v3.0.0 - that's labeled as an 'alpha' release. Maybe try to go to the 'Latest Release'? which is v2.1.0

I'm using v2.1.0 with no issues (starting over 60 slaves on one pc) as long as things are Disposed properly (as @Maxwe11 said) between one startup and the next. For reference, here's the core of my startup code:
`                slaveTcpListener = new TcpListener(ip, TcpPort);

                slaveTcpListener.Start();

                slave = ModbusTcpSlave.CreateTcp(SlaveID, slaveTcpListener);
                slave.DataStore = DataStoreFactory.CreateDefaultDataStore();

                slave.Listen();
`
Hope this helps. :)

AurelienC, 2017-07-24:
Sorry, no much time.

I made a new project with Node.js and, thus, an other package for Modbus communication. :-)

closed 2017-07-24 by AurelienC