NModbus4/NModbus4 #117: Checksum failed / message cut off

everyusernameistaken123, 2016-12-06:
Hi,

For some reason, the "CreateResponse" method does not work for me.

First, this is the error / exception I am getting:
"Checksums failed to match 1, 3, 4, 255, 255 != 1, 3, 4, 255, 255, 255, 255, 251, 167"

I investigated this failure the whole day and I have to believe the problem is located in the "ModbusMessageImpl" class?
The whole byte-array I receive is:
RX: 1, 3, 4, 255, 255, 255, 255, 251, 167
(It is a valid Modbus Response to my initial ReadHoldingsRegister - request)

If this "frame" is used in ModbusSerialTransport.CreateResponse, I get the mentioned exception. The IModbusMessage.messageFrame is for some reason not complete? Only first 5 elements?

Any help is greatly appreciated!

everyusernameistaken123, 2016-12-07:
I am sorry, it was my fault :(
I was rewriting the NModbus4-lib to work without "Array.Slice" because this method is not supported by my framework... And well, I made mistake in doing so :(

Maxwe11, 2016-12-07:
@everyusernameistaken123 which framework? If it's not a secret.

closed 2016-12-07 by Maxwe11

everyusernameistaken123, 2016-12-07:
I am using UWP (Universal Windows platform) as target for my project. This is only using ".NETCore" and not the full .NET-framework.