NModbus4/NModbus4 #118: Run multiple instance with different slave Id's

HarishNayak, 2016-12-12:
Everything working fine, but when I use different slave Id's, it's using some default slave Id,,not able to find out the issue, any suggestions please.

Maxwe11, 2016-12-13:
@HarishNayak for which transport?

HarishNayak, 2016-12-14:
It's for the TCP.

ravigyanvihar, 2017-03-15:
Hi
I am facing same problem. Can you help me out 
in my project I need to read rtu devices connected to gateway(TCP) 
What I did 
I made a TCP client and called readholdingregisters(slave ID, start address, lenghth) 
But it does not work 
Tell me what I need to do more

rquackenbush, 2017-04-11:
I wrote my own RTU slave code to handle just this.  In my case, I had 3 bits:
- ```PromiscuousListener``` (wraps ```IStreamResource```)
- ```ModbusSlaveHost``` - brokers communication between the network and individual slaves
- ```ModbusSlave``` - Multiple instances can be attached to a single ModbusSlaveHost, thus representing multiple slave devices.

Unfortunately, that also meant that I ended up writing function handlers (e.g. ReadMultipleHoldingRegisters) myself.