NModbus4/NModbus4 #119: Dnx Runtime package needs to be installed

MichaelRanftelhuber, 2016-12-22:
Dear nModbus 4 team,

I have been using nModbus 3 for some time and was happy about it and hence wanted to upgrade. So first of all let me say thanks for the good work!

Unfortunately, I cannot build the solution of nModbus 4. Visual Studio reports:

2>C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v14.0\DNX\Microsoft.DNX.targets(126,5): error : The Dnx Runtime package needs to be installed. See output window for more details.

However, if I look in the NuGet Package Manager Console:

```
PM> Install-Package Microsoft.Dnx.Runtime -Pre
Installing NuGet package Microsoft.Dnx.Runtime.1.0.0-rc1-final.
Successfully installed 'Microsoft.Dnx.Runtime 1.0.0-rc1-final' to Samples

PM> dnvm list

Active Version           Runtime Architecture Location                              Alias  
------ -------           ------- ------------ --------                              -----  
       1.0.0-beta8       clr     x64          C:\Users\M.Schmidberger\.dnx\runtimes        
       1.0.0-beta8       coreclr x86          C:\Users\M.Schmidberger\.dnx\runtimes        
  *    1.0.0-rc1-update2 clr     x86          C:\Users\M.Schmidberger\.dnx\runtimes default
```

So it seems that the dnx runtime is installed just fine. I guess there are some broken links or something similar? 

I still am a beginner concerning nModbus, so any help is highly appreciated.

Thanks a million!

Michael

MicheleDAndria, 2017-01-19:
Hi, I'm tring to use nmodbus4 on .net core 1.01 application but i can see some reference on net40 on the nmap4 driver source code. Is it the last version of the library compatible with .net core 1.0.1  without using any addition reference to any assembly?

Maxwe11, 2017-01-24:
@MikeOnTheWork try the latest unstable version

MikeOnTheWork, 2017-01-25:
Hi maxwe11, thanks for your replay. Did you mean  3.0.0-alpha1 release? I can see some references to .net 4.0 framework till. The only project of the solution that looks like a pure .net core 1.0 implementation from an assemblies reference point of view is the project NModbus4.unitTests, may be to it's not clear to me what projects are the ones to be used into my final solution building, could you introduce here what projects of the 3.0.0-alpha1 shall be used in a new solution (what shal be deleted and what shall be used and what references can be deleted that are related only to coveraging and testing).

Thanks