NModbus4/NModbus4 #120: slaveid in TCP

goyzhang, 2017-01-06:
I'm reading data using ModbusTCP with **two** devices connected to a modbus hub in a loop(i.e one tcp port with 2 different slaveid(3 and 4)).Here is my code:

```
tcpclient = new TcpClient("192.168.1.233", 8899);
ModbusIpMaster master = ModbusIpMaster.CreateIp(tcpclient);
while(true)
{
                  ushort startAddress = 4000;

                    ushort numberOfPoints = 4;

                    byte slaveid = 3;      

                    var regs = master.ReadHoldingRegisters(slaveid,startAddress, numberOfPoints);

                 //dosomething

}

```
However I always get this error in my loop:
“{"Response slave address does not match request. Expected 4, received 3."}”
How could it expected 4 ? The slaveid is 3 that I put in. Anyone have the same problem? Or does the slaveid somehow not function under this ModbusTCP setting?

ravigyanvihar, 2017-03-15:
Hi sir Have you find a solution for the problem 
I am facing the same

bkmr, 2017-03-15:
The SlaveId is part of the protocol, and should be checked.

However, each slave will have it's own IP/Port combination, so you will need a separate connection for each slave.

If that doesn't work, you'll need to provide more information about your setup.

goyzhang, 2017-03-17:
@ravigyanvihar Nope, It works unless they have different ip:port combination. I tried a python library called pymodbus on the same setup, and it worked.

@bkmr I have 2 Modbus RTU devices connected to 1 ModBus RTU to TCP hub. The hub contains only one tcp/ip combination which is itself and the 2 devices should be  recognized by their slave id. I.e ,they have the exactly same IP/Port but different unit id. The problem will occur under above situation. 
In Modbus rtu,this can be achieved by COM and slaveid. Modbus Tcp protocol should allow the same ip/port but different unit ids setup for different devices, as this setup works on another python library and can be referenced by Schneider's topology.  As many people are facing the same problem, I think there might be some glitches on Nmodbus4 slave id mechanism.Any idea?

goyzhang, 2020-01-20:
Close this issue since this project is no longer maintained.

closed 2020-01-20 by goyzhang