NModbus4/NModbus4 #122: No content in documentation.

luckluster, 2017-02-20:
There are only page titles in the supplied CHM file, but no content ever shows up.
I am using Windows 10.
Here's a screenshot.

[image]

shaggygi, 2017-02-20:
I've seen this before. [Try this..](http://stackoverflow.com/questions/4400744/chm-viewer-unable-to-show-contents)

"Try to right-click on your chm file and select properties. You will see an "Unblock" button. Click on it, it should solve your problem."

luckluster, 2017-02-20:
Thanks! That helped.

/facepalm

luckluster, 2017-02-22:
After managing to unlock the documentation, it appears that there are no clear instructions on how to use the library.

I know a bit about modbus (through serial connections) and managed to use it, but I can't understand how to create, for example, a modbus slave using the library through a serial connection. 

I looked in the "samples" folder, but that didn't help much (and strangely I had to install .NET-core to make it compile at all)