NModbus4/NModbus4 #124: Any example of ExecuteCustomMessage

ummarbhutta, 2017-03-20:
I need to execute a custom function in user defined function space e.g. 100, I think I think `ModbusMaster.ExecuteCustomMessage<TResponse>(IModbusMessage request)` can be used. I need some example of it, I will appreciate if you can share it with me?

alex-kag, 2017-03-25:
Have ideas how to implement this functionality?

ummarbhutta, 2017-04-03:
@alex-kag Yes I found a way, You need to implement `Modbus.Message.IModbusMessage` for Modbus request and response. I have to figure it out myself how to implement it, as there is no documentation. But once it is implemented it is working now in a great way. Do you need detail implementation for it? I can share an example with you?

alex-kag, 2017-04-03:
Is it possible to get your solution?

jrosa-ak, 2017-04-03:
It would be great to have an example that helps future users as well. If
it's not in the existing documentation perhaps the maintainers will accept
a pull request for it?

Javier Rosa
Software Engineer
jrosa@amberkinetics.com
Office: (510)-474-1000 Ext. 739

Amber Kinetics
32920 Alvarado Niles Blvd #250
Union City, CA 94587

ummarbhutta, 2017-05-05:
There are two typical scenarios

1. Execute a custom message when acting as Modbus Master. This scenario was quite simple and I don't have make any changes in the library and `ExecuteCustomMessage` did all the work.
2. While when we need to act as a Modbus Slave there were some changes required in the library as well to provide an extension point for developer so custom messages can be implemented, I have made those changes as well on a [Fork](https://github.com/ummarbhutta/NModbus4)
3. I will soon update the samples and will checkin + a pull request.

Thanks

ummarbhutta, 2017-05-05:
[Pull request](https://github.com/NModbus4/NModbus4/pull/129) created, Samples projects updated for example. Closing ticket.

closed 2017-05-05 by ummarbhutta