NModbus4/NModbus4 #128: Logging

Pronic32, 2017-04-27:
Is there any way to enable logging in Release assembly? I need to get sent and received data. Or how else could I get errors occurred in production?

rquackenbush, 2017-05-03:
Hi @Pronic32 - I've forked this project to https://github.com/NModbus/NModbus. What kind of error information are you looking for? Are you operating as a Modbus Slave or Master?

Pronic32, 2017-05-03:
Hi @rquackenbush. I'm operating as a Modbus Master. I'm just writing and reading single registers through `ModbusMaster` objects.
I've had a problem - `ModbusMaster` object has written data to registers, but our device has not done what it should have.
The system is in production and I'm trying to find a way for logging the whole messaging exchange, but I've found only `Debug.WriteLine()` calls in the source code.
So I'm looking for a way of logging or for the reason of using `Debug.WriteLine()`.

rquackenbush, 2017-05-04:
I believe that the maintainers of the project ripped out the NLog support a while back and put it simple Debug.WriteLine statements.

The immediately get around this, you could write a simple shim that implements IModbusMaster / wraps ModbusMaster. That way you can intercept and log all of the calls.

In my fork, I've been considering adding a logging component. I think that the trick there will be to pick something that easily integrates with other logging frameworks. Perhaps something like this could be injected into ModbusFactory:

```c#
public interface INModbusLogSink
{
   void Log(Severity severity, string message);
}

public enum Severity
{ 
   None,
   Error,
   Information,
   Debug,
   Verbose
}
```