NModbus4/NModbus4 #131: ReadWriteMultipleRegisters command sends only part of the whole frame

dimaios, 2017-05-17:
I've a problem by using the _ReadWriteMultipleRegisters_ function with Modbus RTU.
After several trials I've investigated on the byte frame transmitted by command FC17 by using a serial line sniffer.

I've compared two libraries under windows OS. 

1. NModbus4 release 2.1.0 ( C# )
2. Libmodus 3.1.4 ( C++ )

Test conditions :

1. USB->RS485 converter : baud rate 1000000 bit/s 
2. Test program in C# ( uses the _ReadWriteMultipleRegisters_ function )
3. Test program in C++ ( uses the _modbus_write_and_read_registers_ function )
4. Same receiver for both tests

RESULTS : 

====>>>>> C# TEST
Sending the command : 

```
try{ 
   ushort[] ud =  _masterRTU.ReadWriteMultipleRegisters( _selectedSlaveAddress , 5 , 1 , 8 , wd ) ;
}
catch( Exception exc)
{
}
```

the sent bytes are :
[image]

======>> C++ TEST
Sending the command : 
`readwriteResult = modbus_write_and_read_registers(mb, 30, 1, tab_reg_write, 20, 1, tab_reg_read);`

the sent bytes are :
[image]

That is correct based on the specification.

Why the NModbus4 command doesn't work and seems to be partial if compared to the libmodbus one?

Thanks in advance for your support.

jacobbenlewis, 2017-12-18:
I just came across this same problem today.  I hadn't used the ReadWriteMultipleRegisters until now, but yeah, only sending the deviceID, the Function Code, and the CRC, 4 bytes.

jacobbenlewis, 2017-12-20:
Pull request #99 addresses this.