NModbus4/NModbus4 #132: Function code 17 not supported

cipriancoj, 2017-05-23:
**Expected behavior:**
To execute the custom message for the function 17.

**Actual behavior:**
The library rises the exception: "Function code 17 not supported" from the ModbusRtuTransport.

**Steps to reproduce the behavior:**
Implement the IModusMessage into ReportSlaveRequest and ReportSlaveResponse, then call:

```
ReportSlaveRequest request = new ReportSlaveRequest(slaveAddress, 17);
ReportSlaveResponse response = modbusMaster.ExecuteCustomMessage<ReportSlaveResponse>(request);

```
**Comments**

My device respond correctly but the library breaks in the switch case, inside the RequestBytesToRead(byte[] frameStart) method inside the ModbusRtuTransport class.

paluneau, 2017-07-03:
I may sugges that you look at my branch of this project: [https://github.com/paluneau/NModbus4](url), as I implemented the Report Slave ID function for RTU communication.

To remove this exception you have to add the function code 17 as a constant in Modbus.cs:
`      
       // supported function codes
        public const byte ReadCoils = 1;
       ...
        public const byte ReadWriteMultipleRegisters = 23;
        public const byte ReportSlaveId = 17;
`

Then, you have to add a case to the switch in ModbusMessageFactory.cs:
`           public static IModbusMessage CreateModbusRequest(byte[] frame)
            {
            ...

            switch (functionCode)
            {
                ...
                case Modbus.WriteMultipleRegisters:
                    request = CreateModbusMessage<WriteMultipleRegistersRequest>(frame);
                    break;
                case Modbus.ReportSlaveId:
                    request = CreateModbusMessage<ReportSlaveIdRequest>(frame);
                    break;
                default:
                    throw new ArgumentException(
                        String.Format(CultureInfo.InvariantCulture, "Unsupported function code {0}", functionCode),
                        "frame");
            }
            return request;
        }`

In your implementation, you will also have to implment responses and requests classes, just like the other modbus functions supported.

masterchiip, 2022-02-19:
A bit late, but thanks a lot 🙏

closed 2022-02-19 by masterchiip