NModbus4/NModbus4 #137: How I can add some character before address in frame format

Wisit, 2017-08-07:
I would like to use NModbus4 to communicate with device as need  frame formate as picture below , I try to find solution to add EF (Hex) before address  , but not fond solution.  

In the picture below  CRC detail is CRC-CCITT  , How I can select properties of NModbus4 for support CRC-CCITT? 

[image]

[image]

Thnks,
Ple.

Wisit, 2017-08-07:
I test with  CRC-CCITTT with  http://sanity-free.org/133/crc_16_ccitt_in_csharp.html  , It work , So you can add it in NModbus4.

61131, 2018-06-30:
@Wisit - This looks to be a Modbus-like protocol, but is not actually Modbus.  While NModbus4 might prove to be a good starting point for your endeavour, I'm not sure if this type of customisation for Modbus-like protocols is necessarily something that wants to be incorporated into this library.

This being said however, from a quick look, one solution may be to add your leading 0xFE byte in the MessageFrame method of ModbusMessageImpl and then handle the CRC calculation and trailing 0x0D/0x0A in a transport class implementation. I suspect however that there will be a number of further changes that will need to be performed for this Modbus-like protocol to be supported.