NModbus4/NModbus4 #139: Write Register without expecting response

jakelstr, 2017-09-01:
Hello,
I have some Modbus RTU devices that can all be wired together. They are setup to simultaneously perform the same task if I write an instruction to address 0x00. I can get them to take this action with NModbus4, however the library always expects to read a response after it writes something to the bus. In this case, none of the slaves can offer a response because they all took action based on the same message. This is by design on the part of the slave manufacturer. Is there a way to setup a special message that will cause the library to not expect a reply? The slave uses the normal Write Multiple Registers function code to perform this function.

61131, 2018-06-30:
@jakelstr - I did a little digging on this request. I understand exactly what you are trying to do and the use-case that you describe is wholly valid.  Unfortunately, the NModbus4 code at the moment does not support any means for "no response" from a Modbus master message - The code in question in ModbusTransport.cs expects a Modbus frame to be returned or an exception to be thrown. While your use-case could be simply handled by throwing a "no error" exception, it is far from a neat model of implementation.  If this is something that is still an issue for you, I can do some further digging to see how I can return a empty response without data without attempting to read data from the input stream or generating an exception.

jakelstr, 2018-06-30:
Thanks for looking into it! At the time I wrote this, my company was having me mockup something which involved this use-case, so I just manually wrote my way around it. They have for the time being dropped this project so I don't personally need it anymore/at this time