NModbus4/NModbus4 #150: Modbus Slave (RTU), how to respond to requests

WattsC-90, 2018-05-30:
Hopefully this is a quick and easy answer/question scenario, but I have been unable to find a definitive sample code snippet to explain how I use the ModbusSlaveRequestReceived event to return registers when a master requests a "ReadHoldingRegisters" function code message.

I can see that the args.Message.FunctionCode will be three (or should be..[untested]) but dont know how to return the register values.

61131, 2018-06-30:
@WattsC-90 - @MPapst has provided the correct answer.  

All register values returned from a ReadHoldingRegisters function code message will return values from it's DataStore.  Depending upon your use-case however, this may not be sufficient (as was the case for me) - Instead of simply returning values from the DataStore, I wanted to instead have call-back functions triggered within my Modbus slave application code.  

Accordingly, I implemented the code in Pull Request #133 which allows for classes to be built upon the DataStore class (or more specifically a newly introduced IDataStore class) that allows me to have more direct access to how my code handles and returns values for received Modbus requests received by a slave instance.  Depending upon your specific needs, this may be something which is of use to you also.

WattsC-90, 2018-07-03:
@61131 this certainly sounds interesting. I have used a library before, which as it turns out, works in the same way but it does seem like to lack the control/visibility from the consuming library. It could also provide different register values based on configurations or many other cases! now you have my mind going!

61131, 2018-07-03:
@WattsC-90 - That is pretty much exactly how I used it. For my purposes, I used this library to provide interface to separate application that was controlling a connected instrument. Through the interface implemented with NModbus4, external process elements could query and control the state of this application and instrument (which didn't otherwise have any industrial process control interface).  

The information and actions made available through this Modbus interface included current state of instrument operations, the starting and stopping of analysis processes and read-back of analysis results performed by the connected instrument.

(Interestingly, the analysis results returned by this instrument were largely floating-point in nature and as such, I needed to split these values across multiple input registers provided by the NModbus4 interface. This allowed these values to be read out as pairs of input registers and then re-packed into meaningful floating-points within the collecting PLC/RTU).

ChangeTheCode, 2021-01-04:
Hi i currently trying to implement an Modbus Slave based on the NModbus library. For me is also not clear how the slave can response to the request which the slave received from an master. 

I got the event `ModbusSlaveRequestReceived` , `DataStoreReadFrom`  but I do not see an opertunity to set the response with the requested bytes. @MPapst  worte it is handled by the datastore but there i can't find any function to set the response which should be sent back to the master. 

Can someone give me a small working example. Scenario: Master requested 2 Registers and those register values are for e.g. random.

 Would be really nice thanks.

MPapst, 2021-01-04:
Why do you want to customize the response?

The NModbus library is doing all the stuff for you. It completely abstracts all the modbus handling you would need to do in any standardized case.
If you want to respond different on a request, you are no longer compliant to the modbus standard.

If your master is sending a request (e.g. Read Holding, Register 0 ... 10), the values of the registers 0 ... 10 from the `HoldingRegisters` Data Collection from the `DataStore` are returned.
If you want to set different values, just put them into the data stores (or more easily write them using Modbus into the datastore). If you want to have a algorithm calculating the data before sending them, you will probably need to implement a DataStore that implements exactly your algorithm.
See here: https://github.com/NModbus4/NModbus4/blob/portable-3.0/NModbus4/Data/DataStore.cs

ChangeTheCode, 2021-01-04:
Okay that makes sense so i have to set the values inside of the datastore and in case those values have change inside of the salve i have to update them in the datasave. 

How can I update a value inside of a register which is a kind of cached in the datasave? To Image such a register holds some measure values like temperature which will be updated by a task which reads those via a serial bus (i2C).

MPapst, 2021-01-04:
Its been a long time since I worked with this library, just because its not .NET Core compatible and you can consider it dead in development.

So you plan should be to have a instance of the datastore you want to write to. Maybe you can inject that one into the Slave when creating it?
When you have got that instance in your i2C reading class, you can manipulate the data in the store.
If that is not possible, simply create a modbus client that connects to localhost and write the data (is really ugly, but as I said, its been a long time for me and this library :))

ChangeTheCode, 2021-01-04:
okay 
do you know a better library which you use and is also .net Core compatible ? I tried to find a better one because the end is to use it on linux. First thought was to get it running and migrate the lib to have the functions which i need so that they can be executed on linux. 

But i am open for better libs

MPapst, 2021-01-04:
I am working with a fork of this one https://github.com/NModbus/NModbus/