NModbus4/NModbus4 #153: License question

dazzalogan, 2018-09-23:
Hi there,

First off - thank you to everyone who is contributing to this much needed library. Excellent work!

Please could you clarify that nmodbus4 is absolutely free to use?
Do users of the library have to reference the library in their documentation anywhere?

Thank you
Regards
Darren

dazzalogan, 2018-09-23:
Hi Gabriel,

Thanks for the link, but I find the jargon somewhat confusing.

In our case we might want to use nmodbus4 within our own software, but it's unclear to me what we then have to pass on to the customer in terms of licensing & openness of source.

Once a parent software contains a MIT license child component, does the MIT license then apply to the whole / parent software.
We have no problem passing on any license text relating to nmodbus4, but would not want to offer our entire software source.

I'm not sure how it works?

Thank you
Regards
Darren

sparkchaser, 2018-10-16:
Here's some information about the MIT license that's written for non-lawyers: [https://tldrlegal.com/license/mit-license](https://tldrlegal.com/license/mit-license)

Basically, you can use the library or the library's source code for whatever you want.  It is not "viral" like the GPL; that is, using the library has no impact on the license you use for the rest of your application.  The only requirement that it places on you is that you have to notify users that the library is licensed under the MIT license.

A simple way to do this would be to include a "thirdparty.txt" file alongside your program's documentation.  That file would simply say something like "This program uses the NModbus4 library version X.X, which is made available under the following license:", then copy-paste the contents of the library's "LICENSE.txt" file.  If you use multiple third-party libraries, you can document them all in this file.

dazzalogan, 2018-10-19:
Thank you, Spark. Most helpful!

closed 2018-10-19 by dazzalogan