NModbus4/NModbus4 #158: How to read a 16 bit register if I'm getting float value in it using NModbus4

samratmatte, 2018-11-23:
I'm using NModbus4 to read a 16-bit register which already has a float value in it. I'm getting this float value directly from respectively relevant software with registers configured for those float values. This register configuration helps me to get recorded value in my console program using NModbus4.

Question is quite simple. Nmodbus returns `ushort[]` for Holding registers and I want to get a float value while reading from response.

Yes, I tried reading two 16-bit registers and fusing them to form float value(32-bit) and NO, I'm not getting desired results.

For example `2.3e-007` results to `0` and number larger than `ushort` datatype results into `32767`. May be I need to change something in library(NModbus4). Just some hint would do the trick. Please can anybody help me?

CDDFERREIRA, 2019-05-27:
NModbus doesn't give you the converter from ushort to floating-point. That is what you probably read as its very typical in the industry. You will need to convert that value. There is multiple ways to do it online, and it all depends on what type of IEEE your device actually replies.
https://gregstoll.com/~gregstoll/floattohex/

hub003, 2019-08-29:
contigugous block to float   

public static float GetFloat(ushort P1, ushort P2)
        {
            int intSign, intSignRest, intExponent, intExponentRest;
            float faResult, faDigit;
            intSign = P1 / 32768;
            intSignRest = P1 % 32768;
            intExponent = intSignRest / 128;
            intExponentRest = intSignRest % 128;
            faDigit = (float)(intExponentRest * 65536 + P2) / 8388608;
            faResult = (float)Math.Pow(-1, intSign) * (float)Math.Pow(2, intExponent - 127) * (faDigit + 1);
            return faResult;
        }