NModbus4/NModbus4 #162: How to read modbus RTU over Ethernet ?

mridah, 2019-02-27:
I'm able to read data from serial port, and Modbus TCP/IP. I have a serial to ethernet converter [Moxa NPort](https://www.moxa.com/en/products/industrial-edge-connectivity/serial-device-servers/general-device-servers/nport-5100-series). It converts serial data to ethernet ( not to Modbus TCP/IP ). 

How can I use NModbus4 to read this data over ethernet ?? @Maxwe11 @breyed @richardlawley @shaggygi  ??

richardlawley, 2019-02-28:
The product page for that device suggests it presents itself as a serial port, so you can ignore the whole ethernet aspect and just read from your local serial port.

CDDFERREIRA, 2019-06-03:
I am looking to perform a similar operation.
I have a device that acts as a gateway converter from Ethernet to RS485 Serial. And, my field device is a Modbus RTU protocol type.
And yes, I can do same as you mentioned my device can act as a simple serial port, if I use the manufacturer serial to Ethernet converter. However, the hardware is also smart enough if am Ethernet connection is open on that specific port to process it internally and redirect to the correct serial port.

I have tested connect the device directly and got it working with NMODBUS4 using the SerialMaster. Now I am looking at interfacing it through my bridge hardware that encapsulates an I.P. address and port number to a serial comm internally.
So its still a RTU Modbus device, but I am using Ethernet to connect to it. Any example help? 

Much appreciated.

ChangeTheCode, 2019-12-26:
I also used such a Modbus RTU to Ethernet converter from Moxa and Moxa provide as tool Name is "Nport" which do the mapping to a COM Port. Like @richardlawley also mentioned. This works fine. 

So for this use case NModbus4 Need no Special Features.

fatihbahceci, 2020-02-11:
@ChangeTheCode  this solution (NPort) is not flexible for dynamic numerous connections