NModbus4/NModbus4 #183: Modbus RTU - ReadHoldingRegisters don't read after 10 seconds

stefurios, 2020-10-14:
I have a problem use this library with Modbus RTU. 
I would like to read the data cyclically after a certain period of time (30 seconds), I have problems if I put the time interval greater than or equal to 10 seconds while I have no problems if this is less. 
In particular I am using the ReadHoldingRegisters and at the second cycle the application gets stuck on the read and I have a substantial increase cpu use. I tried to set all the client properties (ReadTimeout, Retries ...) but the problem persist.
I have not experienced the same problem with Modbus TCP.
Here is a simple example showing the problem.

`            tcpClient = new TcpClient(hostname, port);
            modbusClient = ModbusSerialMaster.CreateRtu(tcpClient);
            modbusClient.Transport.ReadTimeout = 1000;
            modbusClient.Transport.WriteTimeout = 1000;
            
            while (true)
            {
                ushort[] result = modbusClient.ReadHoldingRegisters(slaveAddress, address, numRegister);
                foreach (var item in result)
                {
                    Console.WriteLine(item);
                }
                Thread.Sleep(10000);
            }`

Fonta88, 2020-12-03:
Any news about this issue? I'm facing the same

pvakharwala, 2020-12-05:
Have you tried closing the port after every call?

Fonta88, 2020-12-07:
Which port? It doesn't use a SerialPort (if it's what you mean) for real because it's RTU over TCP.
Maybe you was talking of the TCP connection?

pvakharwala, 2020-12-07:
> Which port? It doesn't use a SerialPort (if it's what you mean) for real because it's RTU over TCP.
> Maybe you was talking of the TCP connection?

Ohh sorry I didn't have a through look at it.. I thought you were using serial port...my bad